kennethjiang/OctoPrint-Anywhere #86: file does not exist.

caninjpn, 2019-06-01:
I installed Octoprint and was trying to install Octoprint Anywhere from the plugin manager but it's gave an error saying

> Invalid requirement: 'pip install "https://github.com/kennethjiang/OctoPrint-Anywhere/archive/master.zip"'
> It looks like a path. File 'pip install "https://github.com/kennethjiang/OctoPrint-Anywhere/archive/master.zip"' does not exist.
> Error!
> Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

I'm not sure what the error is.

kennethjiang, 2019-06-03:
I'm puzzled too. The path indeed was correct. Are you running OctoPrint on a Raspberry Pi or a PC?

caninjpn, 2019-06-04:
> I'm puzzled too. The path indeed was correct. Are you running OctoPrint on a Raspberry Pi or a PC?

I'm running on a Raspberry Pi 3B+.
Aside from setting up I had to install a hotfix so the OctoPrint could communicate with my Monoprice Select Mini. I'm not sure if this will help but that's the only thing I can think of that may cause this problem.

kennethjiang, 2019-06-04:
Can you ssh to your Pi and run `/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/kennethjiang/OctoPrint-Anywhere/archive/master.zip --no-cache-dir --process-dependency-links` manually to see if you are getting the same error?

kennethjiang, 2019-06-27:
Closing this issue since there haven't been any reports recently on this issue. It sounds like a temporary issue during the time when OctoPrint 1.3.11 was officially released.

closed 2019-06-27 by kennethjiang