kennethjiang/OctoPrint-Anywhere #87: NOT in repository anymore ???

jarjartee, 2019-06-03:
OctoPrint-Anywhere not I plugin repo ??
URL not valid ??

Installing plugin...
Installing plugin from https://github.com/kennethjiang/OctoPrint-Anywhere/archive/master.zip...
Installation needs to process external dependencies, that might make it take a bit longer than usual depending on the pip version
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/kennethjiang/OctoPrint-Anywhere/archive/master.zip --no-cache-dir --process-dependency-links
Usage:
/home/pi/oprint/bin/python2 -m pip install [options] <requirement specifier> [package-index-options] ...
/home/pi/oprint/bin/python2 -m pip install [options] -r <requirements file> [package-index-options] ...
/home/pi/oprint/bin/python2 -m pip install [options] [-e] <vcs project url> ...
/home/pi/oprint/bin/python2 -m pip install [options] [-e] <local project path> ...
/home/pi/oprint/bin/python2 -m pip install [options] <archive url/path> ...
no such option: --process-dependency-links
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

kennethjiang, 2019-06-03:
I just tested it and it installed normally on my OctoPi. Can you ssh to your Pi and run `/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/kennethjiang/OctoPrint-Anywhere/archive/master.zip --no-cache-dir --process-dependency-links` manually? Also can you download and post `plugin_pluginmanager_console.log` here?

kennethjiang, 2019-06-04:
@jarjartee  have you got a chance to run the command above?

jarjartee, 2019-06-04:
Thanks, I’ll try tomorrow, had a mad day at work

kennethjiang, 2019-06-27:
There have been any reports of similar issues recently. I'm closing it assuming it was a temporary issue during the time when OctoPrint 1.3.11 was released.

closed 2019-06-27 by kennethjiang