kennethjiang/OctoPrint-Anywhere #88: Waiting for Camera Feed

smiles2102, 2019-06-08:
Hey there,

I have perused the other issues related to this issue, but the fix all seem to be specific to the user.

I am a new user of Octopi (~2 weeks) and installed your plugin, Octoanywhere.  When accessing octoanywhere.io, I receive the "Waiting for Camera Feed" and it never populates. 

I saw that you previously ask for octoprint.log, so here is mine from today. 
[octoprint.log](https://github.com/kennethjiang/OctoPrint-Anywhere/files/3268721/octoprint.log)

Thank you for your help, and let me know if you need anything else.

kennethjiang, 2019-06-09:
From your log file it looks like your Raspberry Pi occasionally loses internet connection but most of the times you should be fine.

Do you see the temperatures and print job status on OctoPrint Anywhere dashboard?

smiles2102, 2019-06-09:
Hi there,

there is potential that was from me turning it off and on again? Or maybe not.

I do see the temperatures and print job status.

[image]

kennethjiang, 2019-06-10:
So the plugin was able to talk to OA server just fine but somehow the image data didn't go through.

To further debug it, I will need to have a video conferencing session with you to go through a few trouble-shooting steps. Can you send email to support@getanywhere.io so that we can schedule a time?

closed 2019-06-27 by kennethjiang

smiles2102, 2019-06-27:
Thanks!  I upgraded to a webcam and it works fine now.

Closed #88<https://github.com/kennethjiang/OctoPrint-Anywhere/issues/88>.