kennethjiang/OctoPrint-Anywhere #90: Not showing up in Plugin Manager or launching Setup Wizard

irmcdermott, 2019-06-18:
I'm brand new, so I apologize if this isn't the right place for this. 

Since I'm new, I've installed the latest version of OctoPi, and I downloaded Octoprint Anywhere from the repository. The on screen dialog says it installed successfully, but the Setup Wizard doesn't open. I tried reinstalling and it tells me that it is already installed but forces a new install. It also is not showing up in my Plugin Manager list, or at the bottom of the left hand column of my settings page with the rest of my installed. Am I doing something wrong?

kennethjiang, 2019-06-18:
Thank you for submitting this issue report and sorry for the problem you are experiencing.

Can you attach "octoprint.log" to this issue so I can take a look? You can find the log file by following this instruction: https://community.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299

irmcdermott, 2019-06-19:
[octoprint.log](https://github.com/kennethjiang/OctoPrint-Anywhere/files/3304278/octoprint.log)

kennethjiang, 2019-06-19:
Thank you @irmcdermott for sending me the log file. The log file revealed a weird error that seems to be caused by failure in python package management system.

Can you ssh to your OctoPi and run this command and post herewhat you see on the screen?

```
/home/pi/oprint/bin/python -m pip install backoff==1.4.3
```

irmcdermott, 2019-06-19:
[op anywhere.txt](https://github.com/kennethjiang/OctoPrint-Anywhere/files/3306046/op.anywhere.txt)

kennethjiang, 2019-06-19:
It looks like python module `backoff` was installed successfully. Can you restart OctoPi and see if OctoPrint Anywhere works correctly now?

irmcdermott, 2019-06-20:
i restarted and it was showing up in my plugin manager, but not showing up on the left hand column underneath "Plugins" and still not getting any kind of Setup Wizard. Sorry! 
[plugin_pluginmanager_console.log](https://github.com/kennethjiang/OctoPrint-Anywhere/files/3308513/plugin_pluginmanager_console.log)
[octoprint.log](https://github.com/kennethjiang/OctoPrint-Anywhere/files/3308514/octoprint.log)
[image]

kennethjiang, 2019-06-20:
Thank you @irmcdermott .

Your new log file shows that it's now failing at another python module. I'm pretty convince that it is caused by a strange python module management bug.

Please ssh to your Pi and run this command `/home/pi/oprint/bin/python -m pip install raven picamera==1.13`. 

These should be the last dependencies that were somehow in corrupted state. After that you restart OctoPi and the OctoPrint Anywhere plugin should work correctly.

NeoRame, 2019-06-20:
Same Problem here. i tried both commands... 
the result till now, i can see the plugin in the manager, but no wizzards starts and there is no plugi in in the plugin list in the bottom left hand  corner.

[plugin_pluginmanager_console.log](https://github.com/kennethjiang/OctoPrint-Anywhere/files/3312243/plugin_pluginmanager_console.log)
[octoprint.log](https://github.com/kennethjiang/OctoPrint-Anywhere/files/3312244/octoprint.log)

NeoRame, 2019-06-20:
THANK You, find a solution. uninstalled raven and reinstalled anywhere as you  mentioned here: https://github.com/kennethjiang/OctoPrint-Anywhere/issues/56#issuecomment-437495086

kennethjiang, 2019-06-20:
Thank you @NeoRame . I already forgot that we ran into similar issues before! Definitely a weird bug in python module management and/or Pi.

irmcdermott, 2019-06-21:
Thank you to both of you! Appears to be up and running now!

closed 2019-06-21 by kennethjiang