kennethjiang/OctoPrint-Anywhere #91: fail to install on none raspberry pi device.

eikeime, 2019-06-18:
Cant successfully install plugins on none raspberry pi device due to picamera. 

I am using a Nanopi Neo as a Octoprint server, when install from control panel, i get Error log:
`Building wheels for collected packages: OctoPrint-Anywhere, picamera
  Building wheel for OctoPrint-Anywhere (setup.py) ... done
  Stored in directory: /tmp/pip-ephem-wheel-cache-4Jav6n/wheels/8b/a8/f7/cbf9348fc046008c688339a2cd21176147e6359ca931635da7
  Building wheel for picamera (setup.py) ... error
  ERROR: Complete output from command /tools/OctoPrint/venv/bin/python -u -c 'import setuptools, tokenize;__file__='"'"'/tmp/pip-install-wnbayc/picamera/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-ivxM_B --python-tag cp27:
  ERROR: running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv7l-2.7
  creating build/lib.linux-armv7l-2.7/picamera
  copying picamera/streams.py -> build/lib.linux-armv7l-2.7/picamera
  copying picamera/renderers.py -> build/lib.linux-armv7l-2.7/picamera
  copying picamera/__init__.py -> build/lib.linux-armv7l-2.7/picamera
  copying picamera/color.py -> build/lib.linux-armv7l-2.7/picamera
  copying picamera/display.py -> build/lib.linux-armv7l-2.7/picamera
  copying picamera/encoders.py -> build/lib.linux-armv7l-2.7/picamera
  copying picamera/array.py -> build/lib.linux-armv7l-2.7/picamera
  copying picamera/bcm_host.py -> build/lib.linux-armv7l-2.7/picamera
  copying picamera/mmal.py -> build/lib.linux-armv7l-2.7/picamera
  copying picamera/frames.py -> build/lib.linux-armv7l-2.7/picamera
  copying picamera/camera.py -> build/lib.linux-armv7l-2.7/picamera
  copying picamera/exc.py -> build/lib.linux-armv7l-2.7/picamera
  copying picamera/mmalobj.py -> build/lib.linux-armv7l-2.7/picamera
  running egg_info
  writing requirements to picamera.egg-info/requires.txt
  writing picamera.egg-info/PKG-INFO
  writing top-level names to picamera.egg-info/top_level.txt
  writing dependency_links to picamera.egg-info/dependency_links.txt
  reading manifest file 'picamera.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  no previously-included directories found matching 'debian'
  no previously-included directories found matching 'docs'
  writing manifest file 'picamera.egg-info/SOURCES.txt'
  installing to build/bdist.linux-armv7l/wheel
  running install
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-install-wnbayc/picamera/setup.py", line 145, in <module>
      main()
    File "/tmp/pip-install-wnbayc/picamera/setup.py", line 140, in main
      cmdclass             = {'install': CustomInstallCommand},
    File "/tools/OctoPrint/venv/local/lib/python2.7/site-packages/setuptools/__init__.py", line 145, in setup
      return distutils.core.setup(**attrs)
    File "/usr/lib/python2.7/distutils/core.py", line 151, in setup
      dist.run_commands()
    File "/usr/lib/python2.7/distutils/dist.py", line 953, in run_commands
      self.run_command(cmd)
    File "/usr/lib/python2.7/distutils/dist.py", line 972, in run_command
      cmd_obj.run()
    File "/tools/OctoPrint/venv/local/lib/python2.7/site-packages/wheel/bdist_wheel.py", line 228, in run
      self.run_command('install')
    File "/usr/lib/python2.7/distutils/cmd.py", line 326, in run_command
      self.distribution.run_command(command)
    File "/usr/lib/python2.7/distutils/dist.py", line 972, in run_command
      cmd_obj.run()
    File "/tmp/pip-install-wnbayc/picamera/setup.py", line 109, in run
      raise ValueError('This system does not appear to be a Raspberry Pi')
  ValueError: This system does not appear to be a Raspberry Pi
  ----------------------------------------
  ERROR: Failed building wheel for picamera
  Running setup.py clean for picamera
Successfully built OctoPrint-Anywhere
Failed to build picamera
Installing collected packages: picamera, OctoPrint-Anywhere
  Running setup.py install for picamera ... error
    ERROR: Complete output from command /tools/OctoPrint/venv/bin/python -u -c 'import setuptools, tokenize;__file__='"'"'/tmp/pip-install-wnbayc/picamera/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-_8_X8P/install-record.txt --single-version-externally-managed --compile --install-headers /tools/OctoPrint/venv/include/site/python2.7/picamera:
    ERROR: running install
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-wnbayc/picamera/setup.py", line 145, in <module>
        main()
      File "/tmp/pip-install-wnbayc/picamera/setup.py", line 140, in main
        cmdclass             = {'install': CustomInstallCommand},
      File "/tools/OctoPrint/venv/local/lib/python2.7/site-packages/setuptools/__init__.py", line 145, in setup
        return distutils.core.setup(**attrs)
      File "/usr/lib/python2.7/distutils/core.py", line 151, in setup
        dist.run_commands()
      File "/usr/lib/python2.7/distutils/dist.py", line 953, in run_commands
        self.run_command(cmd)
      File "/usr/lib/python2.7/distutils/dist.py", line 972, in run_command
        cmd_obj.run()
      File "/tmp/pip-install-wnbayc/picamera/setup.py", line 109, in run
        raise ValueError('This system does not appear to be a Raspberry Pi')
    ValueError: This system does not appear to be a Raspberry Pi
    ----------------------------------------
ERROR: Command "/tools/OctoPrint/venv/bin/python -u -c 'import setuptools, tokenize;__file__='"'"'/tmp/pip-install-wnbayc/picamera/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-_8_X8P/install-record.txt --single-version-externally-managed --compile --install-headers /tools/OctoPrint/venv/include/site/python2.7/picamera" failed with error code 1 in /tmp/pip-install-wnbayc/picamera/
`

kennethjiang, 2019-06-19:
Thank you so much @eikeime for reporting this bug. I don't have a NanoPi so I need your help to figure out where it is broken and how to fix.

Can you ssh to your OctoPi to run these commands and post here the output of these commands?

```
/home/pi/oprint/bin/python -c 'import distutils.util; print(distutils.util.get_platform());'
```

```
/home/pi/oprint/bin/python -c "print(open('/proc/cpuinfo', 'r').read())"
```

eikeime, 2019-06-19:
thank u for your great project first.
my octoprint is located in "/tools/OctoPrint/venv/"
so i run those two command from there, the returned result are 
`linux-armv7l`

and 

`processor	: 0
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 29.71
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 1
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 29.71
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 2
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 29.71
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 3
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 29.71
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

Hardware	: sun8i
Revision	: 0000
Serial		: 02c00081e029938d
`

kennethjiang, 2019-06-20:
It looks like Nano Pi is using exactly the same CPU modules as RPi and that's why the plugin failed to tell them apart. I do a bit more research to see what's the best way to reliably identify if the plugin is running inside a RPi but it may take a while... sorry! @eikeime

thubot, 2019-06-21:
Same problem here on a Orange Pi Zero. Maybe try using

thubot, 2019-06-21:
Same problem here on an Orange Pi Zero.
You should probably not just grep for armv6l/armv7l, because there's a lot of these platforms out there that are not Raspberry Pis ;)
As per [[1]](https://raspberrypi.stackexchange.com/questions/5100/detect-that-a-python-program-is-running-on-the-pi/61071#61071), I'd suggest checking `/sys/firmware/devicetree/base/model` or the `Hardware:` line of `/proc/cpuinfo`

eikeime, 2019-06-21:
no hurry, just grabbed another raspberry pi zero for the project on hand.

kennethjiang, 2019-06-21:
Thank you so much your help and supportiveness. We just fixed it based on @thubot 's suggestion and released patch version 1.0.1. Please uninstall the plugin and install it again and it should work. I'll leave this issue open for a few days in case you still have problems with installation.

eikeime, 2019-06-22:
Just installed it again on the test board, total success, thank u for your quick response on this problem. :D

closed 2019-06-22 by eikeime