kennethjiang/OctoPrint-Anywhere #93: Clean Uninstall or Reinstall?

Quintin9801, 2019-06-26:
You can't make this stuff up, logged into my octoprint after switching from wifi to ethernet and figured the plugin would have to be reconfigured. It ended up disappearing from my plugin manager and I'm looking for a clean uninstall method to reinstall and set this plugin back up. Any help would be greatly appreciated!

kennethjiang, 2019-06-27:
This sounds odd. If it's not in plugin manager, you can also install it as if it was never installed and it should show up (unless there is an installation failure in which case please post the error message here).

Let me know if that works for you.

Quintin9801, 2019-06-27:
After trying to install from the plugin manager, here is the activity on the server. It may be worth noting that I am running octoprint on a windows machine instead of the usual raspberry pi.

`2019-06-27 13:41:37,427   c:\octoprint\venv\scripts\python.exe -m pip --disable-pip-version-check install https://github.com/kennethjiang/OctoPrint-Anywhere/archive/master.zip --no-cache-dir
2019-06-27 13:41:38,622 > Collecting https://github.com/kennethjiang/OctoPrint-Anywhere/archive/master.zip
2019-06-27 13:41:39,658 > Downloading https://github.com/kennethjiang/OctoPrint-Anywhere/archive/master.zip
...
2019-06-27 13:41:43,009 > Building wheels for collected packages: OctoPrint-Anywhere
2019-06-27 13:41:43,009 > Building wheel for OctoPrint-Anywhere (setup.py): started
2019-06-27 13:41:46,029 > Building wheel for OctoPrint-Anywhere (setup.py): finished with status 'done'
2019-06-27 13:41:46,029 > Stored in directory: c:\users\modix1~1\appdata\local\temp\pip-ephem-wheel-cache-ppvlj_\wheels\8b\a8\f7\cbf9348fc046008c688339a2cd21176147e6359ca931635da7
2019-06-27 13:41:46,029 > Successfully built OctoPrint-Anywhere
2019-06-27 13:41:46,032 * Looks like the plugin was already installed. Forcing a reinstall.
2019-06-27 13:41:46,032   c:\octoprint\venv\scripts\python.exe -m pip --disable-pip-version-check install https://github.com/kennethjiang/OctoPrint-Anywhere/archive/master.zip --no-cache-dir --ignore-installed --force-reinstall --no-deps
2019-06-27 13:41:47,230 > Collecting https://github.com/kennethjiang/OctoPrint-Anywhere/archive/master.zip
2019-06-27 13:41:48,230 > Downloading https://github.com/kennethjiang/OctoPrint-Anywhere/archive/master.zip (11.8MB)
2019-06-27 13:41:50,404 > Building wheels for collected packages: OctoPrint-Anywhere
2019-06-27 13:41:50,404 > Building wheel for OctoPrint-Anywhere (setup.py): started
2019-06-27 13:41:53,625 > Building wheel for OctoPrint-Anywhere (setup.py): finished with status 'done'
2019-06-27 13:41:53,625 > Stored in directory: c:\users\modix1~1\appdata\local\temp\pip-ephem-wheel-cache-b5oijj\wheels\8b\a8\f7\cbf9348fc046008c688339a2cd21176147e6359ca931635da7
2019-06-27 13:41:53,627 > Successfully built OctoPrint-Anywhere
2019-06-27 13:41:53,627 > Installing collected packages: OctoPrint-Anywhere
2019-06-27 13:41:53,627 > Successfully installed OctoPrint-Anywhere-1.0.2`

[image]
Octoprint Anywhere is not listed in the plugin manager after the actions above took place

kennethjiang, 2019-06-27:
That's very odd. Can you post the "octoprint.log" file here?

Quintin9801, 2019-06-27:
`2019-06-27 10:21:03,546 - octoprint.startup - INFO - ******************************************************************************
2019-06-27 10:21:03,548 - octoprint.startup - INFO - Starting OctoPrint 1.3.11
2019-06-27 10:21:03,549 - octoprint.startup - INFO - ******************************************************************************
2019-06-27 10:21:04,002 - octoprint.util.connectivity_checker - INFO - Connectivity changed from offline to online
2019-06-27 10:21:05,121 - octoprint.startup - INFO - Blacklist processing done, adding 1 blacklisted plugin versions: roomtemp (any)
2019-06-27 10:21:05,378 - octoprint.plugin.core - INFO - Loading plugins from c:\octoprint\venv\lib\site-packages\octoprint\plugins, C:\Users\Modix 1\AppData\Roaming\OctoPrint\plugins and installed plugin packages...
2019-06-27 10:21:09,614 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin did not pass check, not loading.
2019-06-27 10:21:10,767 - octoprint.plugin.core - ERROR - Error loading plugin anywhere
Traceback (most recent call last):
  File "c:\octoprint\venv\lib\site-packages\octoprint\plugin\core.py", line 863, in _import_plugin
    instance = imp.load_module(module_name, f, filename, description)
  File "c:\octoprint\venv\lib\site-packages\octoprint_anywhere\__init__.py", line 14, in <module>
    from .message_loop import MessageLoop
  File "c:\octoprint\venv\lib\site-packages\octoprint_anywhere\message_loop.py", line 12, in <module>
    from .h264_stream import H264Streamer
  File "c:\octoprint\venv\lib\site-packages\octoprint_anywhere\h264_stream.py", line 23, in <module>
    os.mkdir(TS_TEMP_DIR)
WindowsError: [Error 3] The system cannot find the path specified: '/tmp/octoprintanywhere-ts'
2019-06-27 10:21:12,298 - octoprint.plugins.octoprint.plugins.discovery - INFO - pybonjour is not installed, Zeroconf Discovery won't be available
2019-06-27 10:21:12,434 - octoprint.plugins.errortracking - INFO - Initialized error tracking
2019-06-27 10:21:12,437 - octoprint.plugin.core - INFO - Found 29 plugin(s) providing 29 mixin implementations, 54 hook handlers
2019-06-27 10:21:12,533 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2019-06-27 10:21:12,828 - octoprint.server - INFO - Intermediary server started
2019-06-27 10:21:12,829 - octoprint.plugin.core - INFO - Loading plugins from c:\octoprint\venv\lib\site-packages\octoprint\plugins, C:\Users\Modix 1\AppData\Roaming\OctoPrint\plugins and installed plugin packages...
2019-06-27 10:21:12,836 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin did not pass check, not loading.
2019-06-27 10:21:12,894 - octoprint.plugin.core - ERROR - Error loading plugin anywhere
Traceback (most recent call last):
  File "c:\octoprint\venv\lib\site-packages\octoprint\plugin\core.py", line 863, in _import_plugin
    instance = imp.load_module(module_name, f, filename, description)
  File "c:\octoprint\venv\lib\site-packages\octoprint_anywhere\__init__.py", line 14, in <module>
    from .message_loop import MessageLoop
  File "c:\octoprint\venv\lib\site-packages\octoprint_anywhere\message_loop.py", line 12, in <module>
    from .h264_stream import H264Streamer
  File "c:\octoprint\venv\lib\site-packages\octoprint_anywhere\h264_stream.py", line 23, in <module>
    os.mkdir(TS_TEMP_DIR)
WindowsError: [Error 3] The system cannot find the path specified: '/tmp/octoprintanywhere-ts'`

kennethjiang, 2019-06-27:
It looks like you are running OctoPrint on a Windows machine, but somehow the plugin thought of it as a Pi. Are you running Window on a PC or a Pi (if it's possible).

Quintin9801, 2019-06-27:
I'm running windows on a PC for better performance.

kennethjiang, 2019-06-27:
Will test it when I find a windows machine to see if I can reproduce the same error.

Quintin9801, 2019-06-27:
That would be amazing, Thank you so much for being so quick to respond!

kennethjiang, 2019-06-28:
Just fixed it. If you install it again it should work now. @Quintin9801

Quintin9801, 2019-06-28:
Works great! No problems at all. Thank you so much.

closed 2019-06-28 by Quintin9801