kennethjiang/OctoPrint-Anywhere #95: Setup window pops up every single boot

edfungus, 2019-06-30:
The setup window pops up every single print even though everything is setup. Seems like a bug that isn't tracking if it is setup yet or not.

kennethjiang, 2019-07-01:
We have received multiple reports on this issue since OctoPrint 1.3.11 was released. It seems like a new bug in OctoPrint 1.3.11. 

I have submitted [an OctoPrint issue](https://github.com/foosel/OctoPrint/issues/3211) to report this bug.

For now, you can work around this problem by hard-refreshing your page. Once you have done that, the setup wizard will stop popping up.

kennethjiang, 2019-07-01:
I updated version to 1.0.5 to give users warning when wizard window keeps on re-appearing.

jaspbot, 2021-01-19:
III canttt evenn presss the Finish buttonnn

jaspbot, 2021-01-19:
i have spaghettidetective installed, installed anywhere as well then i reboot system and get the setup screen and cant pres finish.

then i delete anywhere using ssh, i start up again now isee the message about the bug and hard refresh but still cant press finish.

so what next? how can this be fixed? so i cant get it to disapear.. as i cant press finish.. anyone has had this before? thvm.

jaspbot, 2021-01-19:
works now.. nvm think it was buggin when deleted :) so no problems here :)
 
thanks anyway