kennethjiang/OctoPrint-Anywhere #96: Camera feed window in front of all TouchUI interface on Raspberry Pi screen

tjdouble07, 2019-07-01:
Since latest update, my printer screen has the camera feed on top of all TouchUI menus. This is only on my Raspberry Pi backpack screen, not on my phone TouchUI interface or windows computer OctoPrint screen. When I disable OctoPrint Anywhere, the issue resolves.

kennethjiang, 2019-07-02:
When using Rasbperry Pi backpack screen, do you launch a browser directly on Pi?

tjdouble07, 2019-07-02:
Hello,

Thank you for the assistance.

To be honest, I am not 100% sure, as I am pretty new to using Octoprint and
setting up a Raspberry Pi. I followed a few of the old guides on TouchUI
GitHub (not sure which one I got to work), as the new automated
installation didn't want to work with my screen.

https://github.com/BillyBlaze/OctoPrint-TouchUI

https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup#raspberrypi--touchscreen

The screen I'm using is here:

4 inch HDMI LCD IPS Display 800x480 Resolution Resistive Touch Screen HDMI
Interface for Raspberry Pi 3 B/3 B+/2 B/B+/B Zero W
https://www.amazon.com/dp/B07P5H2315/ref=cm_sw_r_cp_apa_i_Lh5gDb9YF655A

Here is what I'm seeing:

https://photos.app.goo.gl/VWrsBa71TsgLUi5F8

Is there anything I can check using your input that would be of assistance,
maybe through Putty?

Sorry for my lack of knowledge with this, as I only just got exposed to
Octoprint and Raspberry Pi a few months ago.

Trevor

kennethjiang, 2019-07-03:
Thank you for providing these details. It does look like a bug in OctoPrint Anywhere plugin. 

Based on the links you shared, it looks like this bug only appears in Chromium browser, which is probably due to the fact that it has a different behavior than other common browsers.

I guess the tricky part is I don't have a LCD display so I can't reproduce the problem myself.

I will think about what I can do. Meanwhile, if anyone with a LCD display can pitch in to help I'll really appreciate it!

kennethjiang, 2019-07-28:
@tjdouble07 This problem is fixed in version 1.0.7. Please upgrade when you get a chance. Feel free to re-open this issue if the problem persists after the upgrade.

closed 2019-07-28 by kennethjiang

tjdouble07, 2019-07-29:
Thank you very much!