kennethjiang/OctoPrint-Anywhere #97: Can't see Anywhere plugin after successfully installed

skylu3d, 2019-07-25:
This is my first time to set up Octoprint ans thing happened to me on OctoPrint: 1.3.11...
I can't see the Anywhere plugin in after I tried some of the suggestion from 
[https://github.com/kennethjiang/OctoPrint-Anywhere/issues/56](url)
and
[https://github.com/kennethjiang/OctoPrint-Anywhere/issues/90](url)

I tried to download the archive and install manually still no luck ;(
Can you help to see anything I missed up would be really appreciate!
[octoprint.log](https://github.com/kennethjiang/OctoPrint-Anywhere/files/3432955/octoprint.log)

kennethjiang, 2019-07-25:
From the log file I think the issue is similar to https://github.com/kennethjiang/OctoPrint-Anywhere/issues/90 but this time it is a different dependency. The fix will be to ssh to your Pi and run:
`/home/pi/oprint/bin/python -m pip install backoff==1.6.0`
Let me know if it'll fix the issue @skylu3d

skylu3d, 2019-07-26:
Thanks my friend @kennethjiang , everything works like charm now!

closed 2019-07-26 by kennethjiang