kennethjiang/OctoPrint-Anywhere #98: Waiting for camera feed - Octoprint camera is working

xxgithubnz, 2019-08-12:
Hi - great plugin!

Unfortunately I'm having an issue with the camera feed. The other stats (eg. hot end and bed temp etc.) are streaming as per normal and I can use the remote but the camera just does not display. 

As a workaround I'm using another plugin https://plugins.octoprint.org/plugins/rtmpstreamer/ which works fine streaming to twitch.io - however I'd rather use a single site for display and remote.

This is a Debian Lite (Buster) install on a rPi 3b+ with Octoprint installed using these instructions: https://community.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspbian/2337

I've attached the log file in the hope that you can help.

Thanks!

[octoprint.log.zip](https://github.com/kennethjiang/OctoPrint-Anywhere/files/3491408/octoprint.log.zip)

kennethjiang, 2019-08-12:
Thank you for attaching octoprint.log with the issue which makes it a lot easier to debug.

The log file reveals a bug in the code. The fix to that bug is included in version 1.0.9 that I just released. You can wait for OctoPrint to prompt you for upgrade, or you can uninstall then re-install OctoPrint Anywhere plugin to get the new version now.

I don't think a lot of users have your setups so you may run into more issues even after upgrading to 1.0.9. Let me know if that's the case.

xxgithubnz, 2019-08-12:
Thanks for the quick reply and analysis!

Before work today I followed your advice and did a quick reinstall of the plugin and can report that it's now working! 

I will do further testing when I get home tonight but in the meantime this is now closed.

closed 2019-08-12 by xxgithubnz

kennethjiang, 2019-08-13:
Good to know everything works (a bit to my surprise as I don't have the same set up to test the plugin)! :)