kennethjiang/OctoPrint-Anywhere #100: Octoprint anywhere "installed unknown" and cannot see in plugins

Scintilladist, 2019-08-28:
Using OctoPrint 1.3.11 and using the plugin manager to install OctoPrint-Anywhere the install goes fine with no errors. When it is done, a dialog comes up saying 'Installed: unknown' and when the reboot of the OctoPrint service is done, the OctoPrint-Anywhere doesn't show up under the tool menu. When I go to install it again, it sees that it was already installed and does a 'force' install, yet it still shows as 'Installed: unknown' and when rebooted, nothing shows up.
[image]

Scintilladist, 2019-08-29:
Yes here they are.
[octoprint.log](https://github.com/kennethjiang/OctoPrint-Anywhere/files/3553477/octoprint.log)
[plugin_pluginmanager_console.log](https://github.com/kennethjiang/OctoPrint-Anywhere/files/3553478/plugin_pluginmanager_console.log)

kennethjiang, 2019-08-29:
It is the same problem as this issue: https://github.com/kennethjiang/OctoPrint-Anywhere/issues/97

Please try the solution in that issue and let me know if it works for you.

Scintilladist, 2019-08-29:
Thank you so much @kennethjiang . It works perfectly now. Sorry about the trouble.

closed 2019-08-29 by Scintilladist