kennethjiang/OctoPrint-Anywhere #102: Version 1.0.13 blocking startup.

JonAugusto, 2019-09-09:
Octoprint Anywhere v:1.0.13 is currently the only 3rd party plugin I have installed and if it is enabled, startup is interrupted and the Octopi reboots in safe mode. Disabling the plugin solves the startup issue. Not sure if it is a conflict with a  first party plugin or not.

kennethjiang, 2019-09-09:
Hmm that sounds bad. Can you post octoprint.log file so that I can take a look? Here is how you find it: https://community.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299

JonAugusto, 2019-09-09:
Hi Kenneth.

It was weird I couldn’t get Octopi to load all morning. I just rebooted the Pi to get a good log file for you and this time it loaded fine. It’s now running without issue.
I’ve attached the log file for what it’s worth but I have no clue why not wouldn’t load all morning and does so now.

Jonathan

kennethjiang, 2019-09-09:
Good to know it's working now

Did you attach the log file to this issue or send it to me by email?

JonAugusto, 2019-09-09:
I sent it via email.

kennethjiang, 2019-09-09:
Did you send it to `support@getanywhere.io` @JonAugusto

JonAugusto, 2019-09-09:
I just attached it to my reply to your email.  I will upload it when I get back to work tomorrow.

kennethjiang, 2019-09-10:
@JonAugusto I didn't see OctoPrint Anywhere errors the log file you sent me that could possibly cause OctoPrint to boot into safe mode. I did see `UNDERVOLTAGE` errors that may have caused your Pi to shut down but I'm not too sure if it was the reason for your OctoPrint to enter safe mode.

I'm closing this issue for now. If the same problem happens again, the best place to ask this question is probably [OctoPrint forum](https://community.octoprint.org/) or  open an issue with OctoPrint.

closed 2019-09-10 by kennethjiang