kennethjiang/OctoPrint-Anywhere #105: No setup wizard

SonBroku, 2019-09-25:
As the title says, I'm not getting a setup wizard as well as it's not showing up in the plugin menu. It shows as installed and enabled, but there's no type of menus or anything for it.

[octoprint(1).log](https://github.com/kennethjiang/OctoPrint-Anywhere/files/3650268/octoprint.1.log)

Here's my log. I already tried several of the other fixes in previous issues like reinstalling raven. I can't seem to figure this out.

kennethjiang, 2019-09-25:
@SonBroku Thank you for reporting the problem.

For your log file it seems that OctoPrint Anywhere's config file is somehow corrupted. You can reset the config file with these steps:

1. SSH to your Raspberry Pi.
2. Delete `/home/pi/.octoprint/data/anywhere/.config.yaml`
3. Restart OctoPrint. Once the OctoPrint restarts you should see the wizard. Let me know if that's not the case.

SonBroku, 2019-09-25:
That worked. I also had installed the polar cloud plugin last night after i made this post, and I had to delete it as well before it would work. Thanks.

closed 2019-09-27 by SonBroku