kennethjiang/OctoPrint-Anywhere #106: Upgrade to 1.0.17 disables the M-JPEG streamer site on port 8080

thesourcerer, 2019-10-05:
Upgraded the plugin to 1.0.17,
after octoprint restart, the pi is no longer responding to port 8080 where the M-JPEG streamer page should display...
Verified by disabling the plugin and restarting octoprint service - everything started working again.

I am willing to repro and provide logs if needed, just let me know what exactly I should fetch.

kennethjiang, 2019-10-05:
Do you have a Pi Camera or a USB Camera.

Please download the `octoprint.log` file and post it here. [Here is how you can find it](https://community.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299).

thesourcerer, 2019-10-05:
Pi Camera

thesourcerer, 2019-10-05:
More info: went back to v .16 and it also has the problem, so I must have skipped a version or more,  going back one by one now to figure out which was the last one to work.

kennethjiang, 2019-10-05:
Ok. Did you upgrade to PRO or sign up for the free trial? The plugin will stop mjpeg streaming so that it can do H.264 streaming. But usually it will also do it in a way that is compatible with the original mjpeg streaming so that you can still see the webcam image in OctoPrint as usually. But something probably went wrong in your case.

Do you see webcam feed on Octoprint Anywhere dashboard?

thesourcerer, 2019-10-05:
Using the free version.
Local stream feed works, only  the streamer page on port 8080 fails (I use it to control the camera).
I didn't try to check if the stream worked on octoprint anywhere. I can check.

kennethjiang, 2019-10-05:
That definitely sounds like a bug. With free version the plugin shouldn't try to mess with the mjpeg at all. Please post the octoprint.log file. Thanks!

thesourcerer, 2019-10-05:
More info; Went back to .15, the issue is still there.
I can see the video feed both locally and in getanywhere.io, but port 8080 is not responding.

thesourcerer, 2019-10-05:
[octoprint.log](https://github.com/kennethjiang/OctoPrint-Anywhere/files/3693226/octoprint.log)

kennethjiang, 2019-10-05:
From the log it looks like you got a temporary error which went away after the last OctoPrint restart. After the restart everything seems to work just fine. Was it what you were seeing, or you are still having issues?

thesourcerer, 2019-10-05:
Nop, the video stream was fixed after the restart, but port 8080 is not accessible unless I disable the plugin. so far I managed to return to .15 with the same issue. didn't have time to go back more now, can try later. It's weird, because I usually update all plugins within 2 weeks tops, and didn't have an issue before.

kennethjiang, 2019-10-05:
Can you send me email `k@getanywhere.io` so that we can schedule a time to have a live chat and walk through some trouble-shooting steps?

thesourcerer, 2019-10-06:
I will , once I can find some free time... I will drop you an email.  maxed out the time I can invest in this this week.
BTW, also noticed that once the plugin is on, even local video stream i slagging, especially at start of the print. not sure why. for now I disabled the plugin and everything seems to work.

kennethjiang, 2019-10-07:
It does sound like the plugin tried to start video streaming. Since you are on a free account, is there any chance that you signed up for the PRO free trial? When you are in free trial, the plugin will try to start video streaming. 
If you have signed up for the PRO free trial, you can give me the email address for your OctoPrint Anywhere and I'll remove the trial for you.

thesourcerer, 2019-10-07:
https://www.getanywhere.io/app/subscription says my current plan is "Free".

kennethjiang, 2019-10-08:
When you are on PRO trial, it'll still say your plan is "free" but you get all the PRO features.

thesourcerer, 2019-10-08:
Emailed you just now.