kennethjiang/OctoPrint-Anywhere #107: Octoprint anywhere doesnt show in plugins

davidjau34, 2019-10-16:
Hello! I trying to install octoprint anywhere on my octo print version 1.3.11 but after installing and the necessaries restart the setup wizard doesn't show up

kennethjiang, 2019-10-16:
Very possibly this is caused by the same issue as https://github.com/kennethjiang/OctoPrint-Anywhere/issues/102 . Can you post `octoprint.log` file here so that I can take a look at it?

davidjau34, 2019-10-16:
[octoprint.log](https://github.com/kennethjiang/OctoPrint-Anywhere/files/3732246/octoprint.log)
There it is

kennethjiang, 2019-10-16:
Confirmed it's the same issue as #102 

The fix is to ssh to your Pi and run:
`/home/pi/oprint/bin/python -m pip install backoff==1.6.0`

It should work after you restart OctoPrint. Let me know if otherwise.

davidjau34, 2019-10-16:
now its show but after a couple of restarts the setup wizard doesn't appear 
[octoprint(1).log](https://github.com/kennethjiang/OctoPrint-Anywhere/files/3732628/octoprint.1.log)

kennethjiang, 2019-10-16:
It looks like in your case there were 2 packages (maybe more) missing. Please run this on your Pi:

`/home/pi/oprint/bin/python -m pip install raven==6.9.0`

davidjau34, 2019-10-16:
It finally works out! thanks, man!

closed 2019-10-16 by davidjau34