kennethjiang/OctoPrint-Anywhere #108: OctoPrint-Anywhere folder on Files

Celtaa, 2019-11-03:
Hi,

Some updates ago I noticed that on Files a folder called "OctoPrint-Anywhere" started appearing. Is this folder necessary? If I try to delete it after some time the folder appears again.
Is there any configuration to disable this?

[image]

kennethjiang, 2019-11-04:
This folder is so that users can upload G-Code and start printing remotely (Pro feature only).

If you don't need such feature, the folder will still be there but stay empty.

Does it create a problem for your situation?

Celtaa, 2019-11-04:
It doesn't create a problem.
For someone that doesn't use the feature or doesn't have pro it doesn't make much sense to have this folder, it is just there creating "noise".

kennethjiang, 2019-11-04:
Gotcha. Since this is not creating a problem, I marked it as enhancement.