kennethjiang/OctoPrint-Anywhere #113: Failure to install on RPi 4B

dchamb55, 2021-01-24:
When I try to install the plugin, this is what is displayed:

Installing plugin "OctoPrint Anywhere" from https://github.com/kennethjiang/OctoPrint-Anywhere/archive/master.zip...
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install file:///tmp/tmpt7AkID/master.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmpt7AkID/master.zip
Requirement already satisfied: OctoPrint>=1.2.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-Anywhere==1.1.1) (1.5.3)
Requirement already satisfied: backoff>=1.4.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-Anywhere==1.1.1) (1.10.0)
Requirement already satisfied: raven in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-Anywhere==1.1.1) (6.10.0)
Collecting picamera==1.13
Downloading https://files.pythonhosted.org/packages/79/c4/80afe871d82ab1d5c9d8f0c0258228a8a0ed96db07a78ef17e7fba12fda8/picamera-1.13.tar.gz (143kB)
Requirement already satisfied: Flask-Babel<2,>=1.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (1.0.0)
Requirement already satisfied: watchdog==0.10.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (0.10.4)
Requirement already satisfied: OctoPrint-FileCheck>=2020.08.07 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (2020.8.7)
Requirement already satisfied: psutil<6,>=5.7 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (5.8.0)
Requirement already satisfied: emoji<1,>=0.5.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (0.5.4)
Requirement already satisfied: monotonic<2,>=1.5; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (1.5)
Requirement already satisfied: filetype<2,>=1.0.7 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (1.0.7)
Requirement already satisfied: PyYAML<6,>=5.3.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (5.4.1)
Requirement already satisfied: feedparser<6,>=5.2.1; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (5.2.1)
Requirement already satisfied: netifaces<1,>=0.10.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (0.10.9)
Requirement already satisfied: future<1,>=0.18.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (0.18.2)
Requirement already satisfied: enum34<1.2,>=1.1.10; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (1.1.10)
Requirement already satisfied: netaddr<1,>=0.7.19 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (0.7.19)
Requirement already satisfied: requests<3,>=2.23.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (2.25.1)
Requirement already satisfied: Click<8,>=7.1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (7.1.2)
Requirement already satisfied: itsdangerous<2,>=1.1.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (1.1.0)
Requirement already satisfied: blinker<2,>=1.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (1.4)
Requirement already satisfied: Jinja2<3,>=2.11.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (2.11.2)
Requirement already satisfied: frozendict<2,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (1.2)
Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (0.5.0)
Requirement already satisfied: flask<2,>=1.1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (1.1.2)
Requirement already satisfied: rsa==4.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (4.0)
Requirement already satisfied: semantic-version<3,>=2.8.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (2.8.5)
Requirement already satisfied: unidecode<0.05,>=0.04.14 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (0.4.21)
Requirement already satisfied: pkginfo<2,>=1.5.0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (1.5.0.1)
Requirement already satisfied: futures<4,>=3.3; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (3.3.0)
Requirement already satisfied: websocket-client<1,>=0.57 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (0.57.0)
Requirement already satisfied: scandir<2,>=1.10; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (1.10.0)
Requirement already satisfied: tornado==5.1.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (5.1.1)
Requirement already satisfied: pyserial<4,>=3.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (3.4)
Requirement already satisfied: markupsafe<2.0,>=1.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (1.1.1)
Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (3.1.1)
Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (2019.8.19)
Requirement already satisfied: chainmap<2,>=1.0.3; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (1.0.3)
Requirement already satisfied: typing<4,>=3.7.4.1; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (3.7.4.1)
Requirement already satisfied: sarge==0.1.5post0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (0.1.5.post0)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2020.09.23 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (2020.9.23)
Requirement already satisfied: sentry-sdk<1,>=0.15.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (0.19.5)
Requirement already satisfied: pylru<2,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (1.2.0)
Requirement already satisfied: werkzeug<2,>=1.0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (1.0.1)
Requirement already satisfied: wrapt<2,>=1.12.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (1.12.1)
Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (2.0)
Requirement already satisfied: cachelib<1,>=0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (0.1)
Requirement already satisfied: contextlib2; python_version < "3.2" in /home/pi/oprint/lib/python2.7/site-packages (from raven->OctoPrint-Anywhere==1.1.1) (0.6.0.post1)
Requirement already satisfied: Babel>=2.3 in /home/pi/oprint/lib/python2.7/site-packages (from Flask-Babel<2,>=1.0->OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (2.7.0)
Requirement already satisfied: pytz in /home/pi/oprint/lib/python2.7/site-packages (from Flask-Babel<2,>=1.0->OctoPrint>=1.2.4->OctoPrint-Anywhere==1.1.1) (2019.3)
[... cut ...] pen)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-lY7CJK/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/oprint/include/site/python2.7/picamera Check the logs for full command output.
Running setup.py install for picamera: finished with status 'error'
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

kennethjiang, 2021-01-24:
Are you using OctoPi 0.18 RC 64-bit build? If so, it's known to have issues.

dchamb55, 2021-01-25:
I am using the following:

OctoPrint 1.5.3 Python 2.7.16 OctoPi 0.17.0 on a Raspberry Pi 4 Model B Rev 1.1

I get notices to upgrade to OctoPi 0.18.0 but have avoided applying the update.

Any suggestions?

Regards

Dale

Dale A. Chamberlain
15210 Humbert Rd
Brighton, IL 62012-1955

Ph. (314) 218-0557
Email: dale@dchamberlain.net

dchamb55, 2021-02-04:
I am using Octopi 0.17.0 64-bit

kennethjiang, 2021-02-19:
Sorry for whatever reason your comments fell through the cracks in my inbox.

I'm pretty sure this is caused by the stability issues with OctoPi 0.18 RC builds. Now OctoPi 0.18 stable build is released. Please re-flash your SD card with the stable build image from https://octoprint.org/download/. Let me know if you still have issues.

dchamb55, 2021-02-19:
I could understand that if I was running Octopi 0.18 but I have never installed it or any RC builds of it. I am running 0.17.

kennethjiang, 2021-02-19:
@dchamb55 I'm under the impression that only 0.18 RC has a 64-bit. 0.17 never did. But I could be wrong.